Armchair-Heavy-Industries/A4T #6: 3mf/Cowlings [Hexagon multi-colour]/Hex A4T Cowling - Revo-Voron - G2-Orbiter spacing [cw2-tap].3mf missing a few lines

mossholderm, 2025-03-02:
Thanks for all the work on A4T!  

It looks like in this file:
3mf/Cowlings [Hexagon multi-colour]/Hex A4T Cowling - Revo-Voron - G2-Orbiter spacing [cw2-tap].3mf

There are a few of the hexagon lines are missing. The indentation for them is present, and shows up in the print, but the "fill" that goes into the indentation is missing. From the front, the missing areas are at 3 0'clock (the partial hexagon along the opening for the fan), and at 4 o'clock (the left hand side of the bottom hexagon).

Thanks again!

mossholderm, 2025-03-02:
Also, I don't know it is just me and/or my version of OrcaSlicer, but there are two large artifacts at the bottom of the 3mf... at least I think they are artifacts.

DW-Tas, 2025-03-07:
Did you say "no" to this prompt when opening in orca slicer?

[image]

And then "yes" to this one?

[image]

It should look something like this after selecting multiple filaments.

[image]

I don't see any artifacts. A screenshot would help :)

mossholderm, 2025-03-08:
That was it.  I was answering "Yes" and then "No". The artifacts that I was seeing were the  missing bits of the hexes. Everything else was scaled correctly except those two small parts. 

Like this:

[image]

closed 2025-03-08 by DW-Tas